henne49/dbus-opendtu #141: 2 opendtu ins VRM integrieren  (HM600 + HMS 2000)

s3sen, 2024-02-14:
Hallo, erstmal vielen Dank für die ganze Mühe, Zeit und die tolle Arbeit.

Aktuell läuft bei mir eine DTU für einen HM600
IP der DTU 192.xxx.xxx.198

nun habe ich wegen dem anderen Funkmodul des HMS 2000 eine weitere DTU mit der IP 192.xxx.xxx.199 aufgesetzt 

beide DTUs sind über ihre  jeweiligen IP Adressen abrufbar und liefern die Daten.

Nun zur Frage: wie bekomme ich die 2te DTU auch ins VRM Portal.

Geht das über ein weiteres Templet ?
Falls ja wie müsste das Templet für 2 DTU Instanzen aussehen? / definiert werden ?

Vorab vielen Dank und Gruß

0x7878, 2024-02-14:
Aktuell eher Copy and paste. Du hast dann zwei Instanzen laufen mit der jeweiligen config. Nenne den zweiten Ordner einfach dbus-dtu2 und starte es.

s3sen, 2024-02-15:
Ok ich versuche es mal, kannst du sagen an welcher Stelle der Befehl geändert werden soll?

henne49, 2024-02-15:
du musst nur den ordner umbenennen, welchen Befehl meinst du?

s3sen, 2024-02-15:
Ich hab versucht es zusätzlichin einen Ordner Data2 zu entpacken:

unzip main.zip "dbus-opendtu-main/*" -d /data2

Das hat nicht geklappt, ich bin echt kein Experte in der Sache. 😉

henne49, 2024-02-15:
nenne den existierenden Ordner um mv alter_name_Ordner neuer_name_Ordner

s3sen, 2024-02-15:
Danke dir, es funktioniert jetzt.
Ich bin begeistert wie schnell man hier Hilfe bekommt. 

Letzte Frage: 
Kann ich im VRM Dashboard die PV Wechselrichter einzeln anzeigen ? 

In der Geräteliste sind beide drin.

Im Dashboard wird nur die Summe angezeigt und auch keine Phase 

[image]

[image]

henne49, 2024-02-15:
Schnell geht es auch nicht immer ;-) im Dashboard sieht man nur die Summary. Da können wir nichts machen.

s3sen, 2024-02-15:
ok, dann ist das so. Das Projekt ist eine extrem coole Sache. Das ermöglicht mir ganz andere Möglichkeiten. Danke 

Ich meine bei einigen im Dashboard die Phasen L1 bis L3 gesehen zu haben. Oder vertue ich mich da gerade ?

FrodoVDR, 2024-02-15:
@s3sen Wenn deine Hoymiles auf verschiedenen Phasen angeschlossen sind und das entsprechend in der OpenDTU konfiguriert wurde, bekommst du diese als L1-L3 angezeigt.
[image]

henne49, 2024-02-15:
Aber nur auf AC IN, wenn es auf AC OUT ist, dass ist es immer auf der Phase des AC OUT

FrodoVDR, 2024-02-15:
Auch bei einem 3 phaseigem System?

s3sen, 2024-02-15:
Danke euch. Find es Mega

henne49, 2024-02-15:
> Auch bei einem 3 phaseigem System? Das ist Schade.

bei einem 3 Phasigen System, kann ich nicht sagen, da ich keines habe. aber beim Einphasigen steht die Phase fest.

closed 2024-02-16 by henne49

s3sen, 2024-02-18:
Hi, ich hab das heute testen können. Auch an AC Out zeigt er die unterschiedlichen Phasen im Dashboard [image]

henne49, 2024-02-18:
Du hast ein 3 oder 1 phasen system?

s3sen, 2024-02-19:
L1 hängt ein HM600 und L2 ein HMS2000, im laufe der Woche kommt auf L3 ein weiterer HM2000. Aktuell ist L3 nicht belegt.
Also alles 1 Phasen WR

henne49, 2024-02-19:
Ist dein Victron System ein 3 Phasen System, nach der Aussage ja, daher kannst du auf AC OUT auch die Phasen sehen? Ansonsten ist die Phase auf AC Out die Phase, auf der der Victron ist.

henne49, 2024-02-19:
Das war ich :-) Unterscheidung Phase geht nur bei einem 3 Phasen Victron an AC OUT und auf allen Phasen auf AC IN egal ob 1 oder 3 Phasen Victron System. Deswegen sehe ich bei meinen 1 Phasen System nicht die Phase auf AC IN, weil die ja festgelegt ist.

s3sen, 2024-02-19:
Achso,verstehe, jedenfalls vielen Dank