henne49/dbus-opendtu #151: get lots of errors, need help with config-file to use for a TASMOTA SML Reader

Jusch777, 2024-02-27:
**Follow the Troubleshooting Section**
Change the Logging Parameter under DEFAULT in /data/dbus-opendtu/config.ini to Logging = DEBUG, please revert, once debugging and troubleshooting is complete. Rerun the script and share the current.log and config.ini file.
[Troubleshooting Section](https://github.com/henne49/dbus-opendtu#troubleshooting )

**Describe the Problem**

I cant get your script running correctly. No new device shows up in the Victron GX devise.

I have a TASMOTA  SML reader on my Energy Meter.
It correctly shows Power, total_in and Total_out. I use it since quite a while with Homeassistant.
The Web-Interface is activated, so with the URL  [http://192.168.2.160/cm?cmnd=status+8](http://192.168.2.160/cm?cmnd=status+8%22)
I get following result json file into the browser:
{"StatusSNS":{"Time":"2024-02-28T00:11:29","SML_RD":{"Power_curr":153,"Total_in":5759.7,"Total_out":52828}}}

**Device you are using**
Tasmota [Tasmota 13.1.0]  which runs on a ESP8266 device.

Steps to reproduce the behavior:
1. Which Venus OS are you using?
     V3.21  (brand new firmware updated)
3. Which Victron Device?
    Victron Multiplus 2 GX  (brandnew)
5. Which DTU?
       what is a DTU ???   
7. Which Template Device?
             Tasmota SML reader 
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[cm.json](https://github.com/henne49/dbus-opendtu/files/14426699/cm.json)
[current.log](https://github.com/henne49/dbus-opendtu/files/14426700/current.log)
[Error-situation.odt](https://github.com/henne49/dbus-opendtu/files/14426701/Error-situation.odt)

**Additional context**
Intention:
The target of this project is to show the Power consumption of my house, collected from the energy meter via the TASMOTA SML reader into the Victron Multiplus GX, to make sure the battery is getting charged/discharged correctly.
This project should be used to use the TSMOTA instead of an extra Victron Hardware-Energy Meter, because there is no space to install one.

Jusch777, 2024-02-28:
WORKING! 
[configini.txt](https://github.com/henne49/dbus-opendtu/files/14429840/configini.txt)
Sorry, forgot the ini  (rennamed to txt since github does not allow to upload ini)

-changed the DTU to template
-changed the sercice name to Servicename=com.victronenergy.acload
-changed your script dbus-opendbu.py to the new servciename   (Line 142)
              servicename = get_config_value(
                config,
                "Servicename",
                "TEMPLATE",
                actual_template,
                "com.victronenergy.acload"

Maybe you can add this to your instruction !

Your project is used to pull in TASMOTA readers by others, like Meine Energiewende](https://www.youtube.com/@MeineEnergiewende) 

I now get the current shown in GX, nevertheless, the over AC-OUT values form Tasmota (IN is shown) is not showing up in the spy report.  The value misssing is the AC-out generated by my PV and supplied into Grid (52828 KWh in the screenshots)
Any Ideas why they are not there ?

[image]

[image]

[image]

henne49, 2024-02-28:
Tasmota is used for PV Inverter and Battery Inverter not GRID, which works. 

Nevertheless, if you want to try
com.victronenergy.acload is wrong for grid also Servicename=com.victronenergy.pinverter which I can see in your config for template0, it should be com.victronenergy.grid

Jusch777, 2024-03-11:
Hi,Sorry ich kann da nicht dranbleiben gerade.Aber nach Umstellung des Service auf .grid.  Funktioniert der Sensor von Dir im GX erst mal prima.Danke dafür !Das Problem: die Regelung im ESS im GX arbeitet so schnell (alle250ms)dass da der externe Sensor vom Stadtwerkezähler (alle 1-4sec)nicht mitkommt.D.h. Der schwingt sich auf.Ich kann jetzt also entweder ein langsamere Regelung selber schreiben und den ESS im GX auf extern stellen, oder sogar die Regelung auf Nulleinspeisung im Homeassistant als Automation erstellen, die alle 10 sec. Nachregelt.Oder natürlich vom Elektriker doch noch eine. Zusätzlichen externen Zähler dazwischenklemen lassen(Anfrage: 600 eur)Was meinst Du ?GrüßeJusch Grüße Jusch

no response, closing

henne49, 2024-03-12:
also ich nutze einen externen Zähler, ich glaube nur das macht wirklich Sinn. 

Aber ich wäre sehr interessiert, wie du das gelöst hast, dass ESS auf den Grid reagiert. Bitte ein paar Screenshost oder eine Anleitung? Das hatte bisher nicht geklappt.