henne49/dbus-opendtu #155: Comment in config.ini is wrong

Sonny13, 2024-03-22:
Hi, thanks for this amazing project.

The documentation states:
AcPosition - > Position shown in Remote Console (0=AC input 1; 1=AC output; 2=AC **input** 2)

But the comment in the config.ini file states:
# AcPosition 0=AC input 1; 1=AC output; 2=AC **output** 2

I checked with my setup and NodeRed. AC input 2 seams to be correct.

henne49, 2024-03-22:
Would you have a photo, where this is loactwd at the multiplus? I ever only used 0 and 1, but never 2

Sonny13, 2024-03-22:
Position Information in Node Red of HMS2000 
[image]

In remote console or vrm the HMS2000 (connected on config 2, and AC out 2) shows up on the grid side of the multiplus, which is wrong.
The HM800 (connected on config 1, and AC out 1) shows up un the correct side.
[image]

henne49, 2024-03-26:
OK thank you for this, we will check this

can you let me know, what node-red package your are using here?

henne49, 2024-04-25:
fixed in https://github.com/henne49/dbus-opendtu/tree/henne49-patch-4

closed 2024-04-25 by henne49