henne49/dbus-opendtu #166: Support Opendtu onbattery

OnkelQuincy, 2024-05-04:
**Follow the Troubleshooting Section**
Change the Logging Parameter under DEFAULT in /data/dbus-opendtu/config.ini to Logging = DEBUG, please revert, once debugging and troubleshooting is complete. Rerun the script and share the current.log and config.ini file.
[Troubleshooting Section](https://github.com/henne49/dbus-opendtu#troubleshooting )

**Describe the Problem**
I use a opendtu onbattery. Two inverters are added. I can´t get connection to Multiplus II 3000 GX with this dbus-opendtu.
Values and opendtu is not shown ind VRM and RemoteConsole.

**Device you are using**
Steps to reproduce the behavior:
1. V3.31
2. Multiplus II 3000 GX
3. OpenDTu onbattery
4. Which Template Device?
SDK v4.4.6-dirty

**Expected behavior**
PV value shown on opendtu onbattery display should be shown in Victron console and VRM

**Screenshots**

[image]
[image]
[image]

**Additional context**
Add any other context about the problem here.

henne49, 2024-05-04:
Please follow the guide: 

Follow the Troubleshooting Section
Change the Logging Parameter under DEFAULT in /data/dbus-opendtu/config.ini to Logging = DEBUG, please revert, once debugging and troubleshooting is complete. Rerun the script and share the current.log and config.ini file.
[Troubleshooting Section](https://github.com/henne49/dbus-opendtu#troubleshooting)

OnkelQuincy, 2024-05-04:
[config.txt](https://github.com/henne49/dbus-opendtu/files/15209866/config.txt)

Where can I find the current.log ?

henne49, 2024-05-04:
Same folder as the config.ini

OnkelQuincy, 2024-05-05:
[image]

Hi, sorry not found. In this folder is no current.log  Could you still take a look at the other information I provided ?

henne49, 2024-05-05:
Have ever done this step? https://github.com/henne49/dbus-opendtu?tab=readme-ov-file#how-to-install

It seems the service is not running at all.

OnkelQuincy, 2024-05-05:
I made it ! Removed everything and startet again. It´s working ! Great job - thanks !!!

[image]

OnkelQuincy, 2024-05-05:
👍

closed 2024-05-05 by OnkelQuincy