henne49/dbus-opendtu #202: OpenDTU inverters not showing up after updates

MichaelJuel, 2024-06-19:
**Follow the Troubleshooting Section**
Change the Logging Parameter under DEFAULT in /data/dbus-opendtu/config.ini to Logging = DEBUG, please revert, once debugging and troubleshooting is complete. Rerun the script and share the current.log and config.ini file.
[Troubleshooting Section](https://github.com/henne49/dbus-opendtu#troubleshooting )

**Describe the Problem**
I have run an OpenDTU (v4.4.3) in the past. Now I have a second (with display and two antennas) with v4.4.7-dirty. Both OpenDTUs working and producing outputs livedata and via 

**Device you are using**
Steps to reproduce the behavior:
1. Which Venus OS are you using? just updated to v3.33
2. Which Victron Device? Cerbo, Multiplus II
3. Which DTU? v4.4.3 and v4.4.7-dirty on different IPs. At the moment both are working. I tried both in the config.ini.
4. Which Template Device? Non. OpenDTU only

**Expected behavior**
The system worked (installantion from about 1 1/2 year ago. System started to behave strange due to log file filled all of space in /data. So I deleted the large log file an updated both the venus and the dbus-opendtu (according to the script in the readme. Running install.sh does not show the inverters in the Cerbo
[config.ini.txt](https://github.com/user-attachments/files/15898818/config.ini.txt)
[current.txt](https://github.com/user-attachments/files/15898820/current.txt)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

0x7878, 2024-06-19:
I'll check this

MichaelJuel, 2024-06-19:
Danke. Gerne auch auf Deutsch.

MichaelJuel, 2024-06-19:
Was mir aufgefallen ist: Die Outputs von api/livedata/status von den beiden DTUs unterscheiden sich im Format. Trotzdem funktioniert es mit beiden nicht.

0x7878, 2024-06-19:
Gerade hängt dbus-opendtu in einer endlosschleife fest, da bei dir python <3.9 läuft. Diese version kennt keinen Listentyp. Du kannst auch die änderung #203  manuell einspielen und schauen was dann passiert.

MichaelJuel, 2024-06-19:
Danke. Ich bin leider zu blöd (kein ITler) um zu verstehen was ich mit dem Hotfix machen muss