henne49/dbus-opendtu #207: The PV Production is counted as consumption

rost1000, 2024-06-28:
**Follow the Troubleshooting Section**
Change the Logging Parameter under DEFAULT in /data/dbus-opendtu/config.ini to Logging = DEBUG, please revert, once debugging and troubleshooting is complete. Rerun the script and share the current.log and config.ini file.
[Troubleshooting Section](https://github.com/henne49/dbus-opendtu#troubleshooting )

**Describe the Problem**
the Victron System show all Pv Inverter as 
Power consumption and Not as production .
Finally it seems to increase the Overall consumption of your System

**Device you are using**
Steps to reproduce the behavior:
1. Which Venus OS are you using? 3.33
2. Which Victron Device? cerboGX
3. Which DTU? V24.3.31
4. Which Template Device?

**Expected behavior**
A clear and concise description of what you expected to happen.
That the Producer power will not increase the sum of the House consumption. 

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
Add any other context about the problem here.

rost1000, 2024-06-28:
Hi,
Yes i have one on AC In and another one on AC Out
[image]

Maybe im not aware if i have to change something else but i have no clue

0x7878, 2024-06-28:
This view does not tell the truth eventually. It is fully controlled by victron. Please check your cerbo view. I guess it can be fixed by trying different AC positions in your config. We have no control over their portal or which data they collect.

rost1000, 2024-06-29:
Hi, just to add this
It's the same in both views
[image]

If both is in AC In it will increase the consumption as well .. i think this is no benefit

0x7878, 2024-06-29:
The problem is still that your inverter is connected to critical load. This is why it is counted as consumption. It cannot flow through the Multiplus to the grid.

rost1000, 2024-06-29:
It doesn't  matter where it is connected .. i have one on each side and both show the same issue.. You can only decect it if the produced amount is higher than the consumption.

seems to be well known but i thought that it's maybe possible to fix it.

It's Unfortunately in german but the same issue
https://community.victronenergy.com/questions/138257/falsches-anzeigen-bei-ac-last.html

closed 2024-06-29 by rost1000