henne49/dbus-opendtu #212: Probleme mit den LOADS

derGreyson, 2024-08-11:
Hi,

ich habe eine Frage, bzw ein "Problem", das meine AC Loads falsch angezeigt werden.

Hier meine Konfiguration:

1. Ich habe einen Hitchi Lesekopf auf meinen Stromzähler per MQTT verbunden. Also richtige Werte
2. Ich habe meine Hoymiles Wechselrichter mit Dbus-opendtu eingebunden.

Jetzt ist es aber so, dass meine Wechselrichter automatisch als + LOADS angezeigt werden.

Damit meine ich Bsp.:
PV Module erzeugen 4000 Watt
Batterie lädt mit 3000 Watt
Zähler zeigt -1000 Watt an
Loads werden mit 3500 Watt angezeigt

Also die Werte stimmen nicht wirklich überein.

[image]

derGreyson, 2024-08-11:
Hier sieht man gut was ich meine.

[image]

derGreyson, 2024-08-11:
Demnach würde ich ja,
4180 Watt einspeisen
und
3405 Watt verbrauchen.
Das wäre ja eine gesamt PV Leistung von 7585 Watt, die ich nicht produziere, bzw produzieren kann.

0x7878, 2024-08-11:
Es ist auch so, das victron blind den Wert vom grid nimmt. Der rechnet da nichts. Victron weiß ebenfalls nicht ob noch weitere PV Inverter/Generator/Wind Turbine am Netz hängen. Auch muss man wissen, dass victron nur korrekt rechnet wenn der Multiplus am L1 hängt. 

Ich verstehe dich und hatte ähnliche bedenken was die Werte anbetrifft daher sollte man es sowieso etwas mit Vorsicht genießen. Am Ende zählt das was dein Zähler bezieht oder einspeist.