henne49/dbus-opendtu #220: No voltage and current at details Venus os 3.50

testpaul999, 2024-11-01:
**Follow the Troubleshooting Section**
Change the Logging Parameter under DEFAULT in /data/dbus-opendtu/config.ini to Logging = DEBUG, please revert, once debugging and troubleshooting is complete. Rerun the script and share the current.log and config.ini file.
[Troubleshooting Section](https://github.com/henne49/dbus-opendtu#troubleshooting )

**Describe the Problem**
No voltage and current at details Venus os 3.50 See 
[image]

**Device you are using**
Steps to reproduce the behavior:
1. Which Venus OS are you using?
3.50
2. Which Victron Device?
RPI with Venus os
3. Which DTU?
See 
[image]

4. Which Template Device?
the first inverter at AC Out of MultiPlus2

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

0x7878, 2024-11-01:
Hi @testpaul999,

Thanks a lot for taking the time to open this issue!

I haven’t had a chance to work with v3.5 myself, so I can’t say for sure if Victron changed something on their end. Right now, everything on our side should be set up correctly, with values sent to the right dbus addresses.

If there’s still a remote console available in v3.5, could you take a look and see if anything’s different? For now, you could try downgrading as a temporary fix or live with this limitation until we find a solution. If it’s something we can address on our end, we’ll definitely work toward a fix.

Thanks again for bringing this to our attention!

0x7878, 2024-11-01:
> Your status is wrong, we fixed that some weeks ago

Sorry, I might have missed this. I still live in the past

testpaul999, 2024-11-01:
> Please update to the latest version, it should work. I have tested it with 3.50
> 
> https://github.com/henne49/dbus-opendtu?tab=readme-ov-file#update-the-code

Hi,  have done this and will have a look if the sun ist out tomorrow. :-)
I‘ll give feedback.

Thx

testpaul999, 2024-11-02:
Gooe mornig from germany,

so there is no change at the display :-( on voltage and current after renew the installation.
[image]
[image]

regards

testpaul999, 2024-11-02:
> Is that the same FW version? With the old GUI?

Yes I’f only change the UI http call from
http://192.168.2.220/gui-v2/
to
http://192.168.2.220/gui-v1/

testpaul999, 2024-11-02:
> Can you also please check the VRM with the new UI? As it works on my system with 3.50

At VRM I can see the voltage and current of the whole day. 👍🏻
I only not showing at the new Web UI.

Regards