henne49/dbus-opendtu #238: Inverters are not shown in venus.OS

NotToBeTold, 2024-12-27:
Describe the Problem
The inverters are not displayed in the system. The openDTU is reachable. Unfortunately, I am not very experienced with this topic and may need some more detailed assistance.

Device you are using
Steps to reproduce the behavior:

Which Venus OS are you using? 3.52
Which Victron Device (Cerbo, Raspi, Multiplus II GX...)? CerboGX
Which DTU (ahoy/opendtu/opendtuonbattery)? openDTU
Which Template Device (Shelly, tasmota,...)? none
Expected behavior
I have modified the config.ini file and attached it to this report. However, no PV inverter is displayed in the system. I have also attached the debugger log for further analysis.

I would greatly appreciate a response and would like to thank you in advance for your assistance!

[image]
[config.txt](https://github.com/user-attachments/files/18263703/config.txt)
[current.txt](https://github.com/user-attachments/files/18263704/current.txt)
[image]
[image]
[image]

NotToBeTold, 2024-12-28:
Good Morning. 
Thank you very much for your fast response! I hope this is helpful
[image]

NotToBeTold, 2024-12-28:
Thank you very much. What stupid mistake :)
[image]

NotToBeTold, 2024-12-29:
Dear Henne49,

First of all, I want to thank you again for the excellent help you provided yesterday. It was incredibly appreciated!

Unfortunately, I’m reaching out again because I’m facing new issues with the setup. Yesterday, it seemed to be working fine, but I noticed that fixed values were displayed (211W for East and 299W for West), and these remained unchanged even overnight, leading me to believe the values were incorrect.

Additionally, one inverter was displayed under PV-Inverter, while the other was shown under Critical Load in the diagram within Venus OS. Now, however, neither of the inverters is showing up anymore.

I have attached the log file and the configuration file once again for your review. Could you please check if there are any inconsistencies in the configuration file? Additionally, are the VRM IDs being used correctly, or could there be conflicts?

I would greatly appreciate your support once again in resolving this problem. Please let me know if you need any further details from my side.

Thank you so much for your time and help!

Best regards,
NotToBeTold
[config.txt](https://github.com/user-attachments/files/18270066/config.txt)
[current.txt](https://github.com/user-attachments/files/18270067/current.txt)
[image]
[image]
[image]
[image]

0x7878, 2024-12-29:
Just a note: in your config are also some spaces after and before the equal sign. I'm not sure how that will be handled.

NotToBeTold, 2024-12-30:
Hi Henne49,

Thanks for your feedback. To clarify:

One inverter is connected as a 3-phase system, and the other is single-phase. So far, everything seems correct in this regard.
The Victron MultiPlus is still in its packaging, so it isn’t measuring any currents yet. Therefore, the split display isn’t a coincidence.
Shouldn’t all inverters simply be aggregated together, with their power output adjusted according to their phase configuration, and displayed as a single cumulative inverter?

Let me know if there’s anything else I should check or adjust.

Kind regards

0x7878, 2024-12-30:
One of your inverters is registered as an inverter instead of an pvinverter. Shout should adjust this in your config.
Servicename=com.victronenergy.inverter
Either remove the line or set it to (I think) pvinverter instead

NotToBeTold, 2024-12-30:
Hi 0x7878,  

Thank you so much for your help! I tried changing the service name, but then nothing was displayed anymore. So, I ended up commenting out the service name, and now both inverters are shown aggregated. Thanks a lot for your support!  

It’s truly amazing that you make something like this available and provide such great support. Can I buy you a beer? 🍺  

P.S. Four Hoymiles are about to join the OpenDTU setup soon – just a heads-up that I might need your help again. 😉  
[image]

Best regards!