henne49/dbus-opendtu #241: Venos OS erhät keine Daten

Oxbow6, 2025-02-09:
Hallo mein Venus Os 3.52 bekommt keine Daten mehr rein vom Ahoy System
Siehe Printscreens
Aber die Ahoy funktioniert soweit
Wenn ich auch im Dbus Spy schaue habe ich von den Wechselrichten keine Daten
Das ganze hat nach einen Reboot nicht mehr funktioniert.
Habt Ihr mir einen Typ ?
Gruss 
Martin

[image]
[image]
[image]

[current.txt](https://github.com/user-attachments/files/18724060/current.txt)

Oxbow6, 2025-02-09:
Hallo habe beim Venus den aktuellen Release 3.54 und alles andere ist auch akuell.
Habe heute auch noch auf V2.0 geupdatet.
Aber leider bekomme ich immer noch keine Werte angezeigt
Wo kann das Problem denn liegen ?

Oxbow6, 2025-02-09:
Also die Version wird mit Version: 1.2.40 angegeben

Oxbow6, 2025-02-09:
So habe nun nochmals das update durchgeführt , nun habe ich die Version 2.0.2 , aber das Problem besteht immer noch

Oxbow6, 2025-02-09:
Ahoy ist die Version 0.8.140
Habe auch schon einen kompletten reinstall durchgeführt

[Json.txt](https://github.com/user-attachments/files/18724265/Json.txt)

[config.txt](https://github.com/user-attachments/files/18724272/config.txt)

Oxbow6, 2025-02-09:
hier noch das Log

[config.txt](https://github.com/user-attachments/files/18724354/config.txt)

Oxbow6, 2025-02-09:
Hallo 
Das Venus OS hat nach dem Booten die Zeit um 10 min verschoben, das war das Problem. Nun läuft alles wieder :-)
Kann man auf dem Venus OS keinen NTP Server hinterlegen ?
Vielen Dank auf da wäre ich nie gekommen

[image]

21-Jailbraik-21, 2025-02-09:
Moin, bin nicht ins der lage zu finden, welche Version ich installiert habe von dbus.
Kann mir jemand helfen?

Danke :)