henne49/dbus-opendtu #245: Nach openDTU Integration wird in Venus OS eine falsche Last berechnet

karmo62, 2025-03-24:
Seit der Integration der openDTU in Venus OS wird die AC Last falsch berechnet.
Es wird wohl die PV Inverter (HM700) Einspeisung als Last dazu gerechnet.
Der Hoymiles speist in ein eigenständiges Hausnetz ein.
Die Anlage mit dem Venus OS ist eine Off-Grid Installation.

Hab den Hoymiles eingebunden um ihm auch über VRM sehen/kontrollieren zu können.

[image]

[image]

Was ist noch falsch konfiguriert?

karmo62, 2025-03-24:
Danke für die schnelle Rückmeldung - bin Anfänger in dem Thema ;-)

> Bitte das Issue Template auch ausfüllen, es fehlen wichtige Informationen, so können wir nicht helfen.
> 
> **Device you are using** Steps to reproduce the behavior:
> 
> 1. Which Venus OS are you using 3.50? - **Verion 3.30**
> 2. Which Victron Device (Cerbo, Raspi, Multiplus II GX...)? - **Raspi**
> 3. Which DTU (ahoy/opendtu/opendtuonbattery) - **openDTU (ESP32**)
> 4. Which Template Device (Shelly, tasmota,...)? - **KEIN Template Device**

karmo62, 2025-03-24:
[image]

karmo62, 2025-03-25:
> Ich sehe da ist kein Multiplus oder Inverter von Victron im Spiel, das ist ein Problem und wurde öfter berichtet. Es funktioniert nur sauber mit einem Victron Inverter und noicht mit einem Venus OS alleine.
>
> Ich sehe da ist kein Multiplus oder Inverter von Victron im Spiel, das ist ein Problem und wurde öfter berichtet. Es funktioniert nur sauber mit einem Victron Inverter und noicht mit einem Venus OS alleine.

Schönen Tag,
mir ist erst im nachhinein auf gefallen, dass der "Phoenix Inverter 12V 800VA" doch ein Victon Inverter ist.
Zählt dieser ned?

k a r l

0x7878, 2025-03-25:
Dein Inverter ist kein Multiplus. Das kann zum problem führen. Du kannst lediglich per config den pvinverter auf ac2 oder input legen und schauen wie die Berechnungen da aussehen. Wir machen hier nichts anderes außer victron zu sagen: hey, wir haben hier einen pvinverter und der ist an acX angeschlossen. Der Rest entscheidet victron. Wir Liefern nur die Werte und rechnen nichts.

karmo62, 2025-03-25:
Verstehe ich es nun richtig, dass die openDTU mit einen Multiplus verbunden die korrekten Werte liefert.
Ausgelesen von einem Hoymiles jedoch falsche Werte anzeigt.