henne49/dbus-opendtu #248: Victron GX und openDTU (dbus-opendtu)

mfreye, 2025-04-22:
Hallo.

openDTU Version: v25.4.10 (per LAN verbunden)
VictronGX Version: beta 3.60~60

Ich habe seit ca. 1Jahr Victron mit openDTu (dbus-opendtu von henne49) am laufen ohne Probleme und immer mit den neusten FW von Victron oder openDTU.

Vor ca. 14 Tagen haben ich dann meine openDTU vom Strom genommen, damit ich meine Hoymiles HMS über Lite-S auf die neueste Version updaten kann.

Danach openDTU an und alle WR haben sich angemeldet.

Dann viel mir auf das die WR nicht mehr im Portal sind (dort wird mir auch 14Tage angezeigt).
Es kann also schon vorher nicht gelaufen sein.

Also geschaut was los ist, aber ohne Erfolg.

Was habe ich gemacht.

Ältere Version von openDTU und GX (Die vorherige FW die auch der GX ist) getestet
…
…
…
dbus-opendtu neu installiert (versucht :wink: )

Also das Problem ist, das ich hier nicht weiter komme:

    root@einstein:~# /data/dbus-opendtu/install.sh
    grep: version.txt: No such file or directory
    grep: version.txt: No such file or directory
    ln: /service/dbus-opendtu/service: File exists
    root@einstein:~# svcadm enable /service/dbus-opendtu
    -sh: svcadm: command not found

Was mich auch wunder ist, das ich kein opkg update machen kann.>

    root@einstein:~# su -
    root@einstein:~# opkg update
    Downloading https://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/all/Packages.gz.

        opkg_download_backend: Failed to download xxxx://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/all/Packages.gz, wget returned 8.
        Downloading xxxx://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/cortexa7hf-neon-vfpv4/Packages.gz.
        opkg_download_backend: Failed to download xxxx://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/cortexa7hf-neon-vfpv4/Packages.gz, wget returned 8.
        Downloading xxxx://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/einstein/Packages.gz.
        opkg_download_backend: Failed to download xxxx://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/einstein/Packages.gz, wget returned 8.

xxxx = https

Sagt einfach was Ihr an Infos braucht.

Danke.

mfreye, 2025-04-23:
Das geht leider auch nicht.

```
root@einstein:~# cp /data/dbus-opendtu/config.ini /data/dbus-opendtu/config.backup
root@einstein:~# wget -O main.zip https://github.com/henne49/dbus-opendtu/archive/refs/tags/$(curl -s https://api.github.com/repos/henne49/dbus-opendtu/releases/latest | grep "tag_name" | cut -d '"' -f 4).zip
--2025-04-23 14:31:04--  https://github.com/henne49/dbus-opendtu/archive/refs/tags/v2.0.2.zip
Resolving github.com... 140.82.121.3
Connecting to github.com|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/henne49/dbus-opendtu/zip/refs/tags/v2.0.2 [following]
--2025-04-23 14:31:05--  https://codeload.github.com/henne49/dbus-opendtu/zip/refs/tags/v2.0.2
Resolving codeload.github.com... 140.82.121.9
Connecting to codeload.github.com|140.82.121.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘main.zip’

main.zip                                                        [ <=>                                                                                                                                      ]  37.58K  --.-KB/s    in 0.06s

2025-04-23 14:31:06 (626 KB/s) - ‘main.zip’ saved [38477]

root@einstein:~# mkdir temp_dir
mkdir: can't create directory 'temp_dir': File exists
root@einstein:~# set -e
root@einstein:~# unzip main.zip -d temp_dir
Archive:  main.zip
replace dbus-opendtu-2.0.2/.coveragerc? [y]es, [n]o, [A]ll, [N]one, [r]ename: error: invalid response [m]
replace dbus-opendtu-2.0.2/.coveragerc? [y]es, [n]o, [A]ll, [N]one, [r]ename: error: invalid response [c]
replace dbus-opendtu-2.0.2/.coveragerc? [y]es, [n]o, [A]ll, [N]one, [r]ename: error: invalid response [c]
replace dbus-opendtu-2.0.2/.coveragerc? [y]es, [n]o, [A]ll, [N]one, [r]ename: error: invalid response [/]
replace dbus-opendtu-2.0.2/.coveragerc? [y]es, [n]o, [A]ll, [N]one, [r]ename: error: invalid response [/]
replace dbus-opendtu-2.0.2/.coveragerc? [y]es, [n]o, [A]ll, [N]one, [r]ename: new name: y
  inflating: rm -r /data/dbus-opendtu-main/y
replace dbus-opendtu-2.0.2/.gitattributes? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/.gitattributes
replace dbus-opendtu-2.0.2/GitVersion.yml? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/GitVersion.yml
replace dbus-opendtu-2.0.2/LICENSE? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/LICENSE
replace dbus-opendtu-2.0.2/config.example? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/config.example
replace dbus-opendtu-2.0.2/constants.py? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/constants.py
replace dbus-opendtu-2.0.2/dbus_opendtu.py? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/dbus_opendtu.py
replace dbus-opendtu-2.0.2/dbus_service.py? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/dbus_service.py
replace dbus-opendtu-2.0.2/docs/ahoy_0.5.93_inverter-id-0.json? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.5.93_inverter-id-0.json
replace dbus-opendtu-2.0.2/docs/ahoy_0.5.93_inverter-id-1.json? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.5.93_inverter-id-1.json
replace dbus-opendtu-2.0.2/docs/ahoy_0.5.93_live.json? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.5.93_live.json
replace dbus-opendtu-2.0.2/docs/ahoy_0.5.93_record-live.json? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.5.93_record-live.json
replace dbus-opendtu-2.0.2/docs/ahoy_0.6.9_live.json? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.6.9_live.json
replace dbus-opendtu-2.0.2/docs/ahoy_0.7.36_live_gap_in_inverter_sequence.json? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.7.36_live_gap_in_inverter_sequence.json
replace dbus-opendtu-2.0.2/docs/ahoy_pre_0.5.93.json? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/docs/ahoy_pre_0.5.93.json
replace dbus-opendtu-2.0.2/docs/opendtu_status.json? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/docs/opendtu_status.json
replace dbus-opendtu-2.0.2/docs/opendtu_v24.2.12_inverter.json? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/docs/opendtu_v24.2.12_inverter.json
replace dbus-opendtu-2.0.2/docs/opendtu_v24.2.12_livedata_status.json? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/docs/opendtu_v24.2.12_livedata_status.json
replace dbus-opendtu-2.0.2/docs/tasmota_shelly_2pm.json? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/docs/tasmota_shelly_2pm.json
replace dbus-opendtu-2.0.2/helpers.py? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
[... 6 lines cut ...]
  inflating: dbus-opendtu-2.0.2/restart.sh
replace dbus-opendtu-2.0.2/run_coverage.sh? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/run_coverage.sh
replace dbus-opendtu-2.0.2/service/log/run? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/service/log/run
replace dbus-opendtu-2.0.2/service/run? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/service/run
replace dbus-opendtu-2.0.2/tests.py? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/tests.py
[... cut ...]
replace dbus-opendtu-2.0.2/tests/test_helpers.py? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/tests/test_helpers.py
replace dbus-opendtu-2.0.2/uninstall.sh? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/uninstall.sh
replace dbus-opendtu-2.0.2/update.sh? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/update.sh
replace dbus-opendtu-2.0.2/version.txt? [y]es, [n]o, [A]ll, [N]one, [r]ename: y
  inflating: dbus-opendtu-2.0.2/version.txt

```

mfreye, 2025-04-23:
Werd ich machen.

Ist das normal?

```
root@einstein:~# /data/dbus-opendtu/install.sh
grep: version.txt: No such file or directory
grep: version.txt: No such file or directory
```

mfreye, 2025-04-25:
1.
```
root@einstein:~# /data/dbus-opendtu/uninstall.sh
Do you really wish to delete the log folder /var/log/dbus-opendtu? [y/n]y
/var/log/dbus-opendtu is deleted
```

2. 
temp_dir 		gelösch per sftp
dbus-opendtu 		gelöscht per sftp

3.
```
root@einstein:~# reboot

Broadcast message from root@einstein (pts/0) (Fri Apr 25 12:38:53 2025):

The system is going down for reboot NOW!
```

4.
```
root@einstein:~# wget -O main.zip https://github.com/henne49/dbus-opendtu/archiv                                                                        e/refs/tags/$(curl -s https://api.github.com/repos/henne49/dbus-opendtu/releases                                                                            /latest | grep "tag_name" | cut -d '"' -f 4).zip
--2025-04-25 12:41:32--  https://github.com/henne49/dbus-opendtu/archive/refs/ta                                                                            gs/v2.0.2.zip
Resolving github.com... 140.82.121.3
Connecting to github.com|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/henne49/dbus-opendtu/zip/refs/tags/v2.0.2                                                                             [following]
--2025-04-25 12:41:33--  https://codeload.github.com/henne49/dbus-opendtu/zip/re                                                                            fs/tags/v2.0.2
Resolving codeload.github.com... 140.82.121.9
Connecting to codeload.github.com|140.82.121.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘main.zip’

main.zip                [ <=>                ]  37.58K  --.-KB/s    in 0.06s

2025-04-25 12:41:33 (681 KB/s) - ‘main.zip’ saved [38477]

root@einstein:~# mkdir temp_dir
root@einstein:~# unzip main.zip -d temp_dir
Archive:  main.zip
   creating: dbus-opendtu-2.0.2/
  inflating: dbus-opendtu-2.0.2/.coveragerc
  inflating: dbus-opendtu-2.0.2/.gitattributes
  inflating: dbus-opendtu-2.0.2/GitVersion.yml
  inflating: dbus-opendtu-2.0.2/LICENSE
  inflating: dbus-opendtu-2.0.2/config.example
  inflating: dbus-opendtu-2.0.2/constants.py
  inflating: dbus-opendtu-2.0.2/dbus_opendtu.py
  inflating: dbus-opendtu-2.0.2/dbus_service.py
   creating: dbus-opendtu-2.0.2/docs/
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.5.93_inverter-id-0.json
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.5.93_inverter-id-1.json
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.5.93_live.json
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.5.93_record-live.json
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.6.9_live.json
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.7.36_live_gap_in_inverter_sequence.j                                                                            son
  inflating: dbus-opendtu-2.0.2/docs/ahoy_pre_0.5.93.json
  inflating: dbus-opendtu-2.0.2/docs/opendtu_status.json
  inflating: dbus-opendtu-2.0.2/docs/opendtu_v24.2.12_inverter.json
  inflating: dbus-opendtu-2.0.2/docs/opendtu_v24.2.12_livedata_status.json
  inflating: dbus-opendtu-2.0.2/docs/tasmota_shelly_2pm.json
  inflating: dbus-opendtu-2.0.2/helpers.py
  inflating: dbus-opendtu-2.0.2/imports.py
  inflating: dbus-opendtu-2.0.2/install.sh
  inflating: dbus-opendtu-2.0.2/restart.sh
  inflating: dbus-opendtu-2.0.2/run_coverage.sh
   creating: dbus-opendtu-2.0.2/service/
   creating: dbus-opendtu-2.0.2/service/log/
[... 38 lines cut ...]
Ergebnis:
com.victronenergy.adc
com.victronenergy.battery.socketcan_can1                                                                                                   Pylontech battery
com.victronenergy.digitalinput.input_1                                                                                                               DI 1 0%
com.victronenergy.digitalinput.input_2                                                                                                              DI 2 30%
com.victronenergy.digitalinput.input_3                                                                                                              DI 3 60%
com.victronenergy.digitalinput.input_4                                                                                                             DI 4 100%
com.victronenergy.digitalinputs
com.victronenergy.fronius
com.victronenergy.grid.cg_BW08800170014                                                                                                    EM24 Energy Meter
com.victronenergy.hub4
com.victronenergy.logger
com.victronenergy.modbusclient.tcp
com.victronenergy.modbustcp
com.victronenergy.platform                                                                                                                             Venus
com.victronenergy.pvinverter.pv_232_2173178                                                                                           Solar Fronius Hausdach
com.victronenergy.settings
com.victronenergy.solarcharger.ttyS6                                                                                                       Solar MPPT 150/70
com.victronenergy.solarcharger.ttyS7                                                                                                      Solar MPPT 250/100
com.victronenergy.system
com.victronenergy.vebus.ttyS4                                                                                                    MP2 Wechselrichter 3x3000Wh

Leider war das alles ohne Erfolg.
Ich habe das so schon auf 4 Systemen installiert.
Ich verstehe es nicht :(

mfreye, 2025-04-25:
nochmal einen reboot gemacht

nochmal ein Update gemacht

```
root@einstein:~# cp /data/dbus-opendtu/config.ini /data/dbus-opendtu/config.backup
root@einstein:~# wget -O main.zip https://github.com/henne49/dbus-opendtu/archive/refs/tags/$(curl -s https://api.github.com/repos/henne49/dbus-opendtu/releases/latest | grep "tag_name" | cut -d '"' -f 4).zip
--2025-04-25 12:49:42--  https://github.com/henne49/dbus-opendtu/archive/refs/tags/v2.0.2.zip
Resolving github.com... 140.82.121.3
Connecting to github.com|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/henne49/dbus-opendtu/zip/refs/tags/v2.0.2 [following]
--2025-04-25 12:49:43--  https://codeload.github.com/henne49/dbus-opendtu/zip/refs/tags/v2.0.2
Resolving codeload.github.com... 140.82.121.9
Connecting to codeload.github.com|140.82.121.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘main.zip’

main.zip                                     [ <=>                                                                               ]  37.58K  --.-KB/s    in 0.04s

2025-04-25 12:49:43 (964 KB/s) - ‘main.zip’ saved [38477]

root@einstein:~# mkdir temp_dir
mkdir: can't create directory 'temp_dir': File exists
root@einstein:~# set -e
root@einstein:~# unzip main.zip -d temp_dir
Archive:  main.zip
   creating: dbus-opendtu-2.0.2/
  inflating: dbus-opendtu-2.0.2/.coveragerc
  inflating: dbus-opendtu-2.0.2/.gitattributes
  inflating: dbus-opendtu-2.0.2/GitVersion.yml
  inflating: dbus-opendtu-2.0.2/LICENSE
  inflating: dbus-opendtu-2.0.2/config.example
  inflating: dbus-opendtu-2.0.2/constants.py
  inflating: dbus-opendtu-2.0.2/dbus_opendtu.py
  inflating: dbus-opendtu-2.0.2/dbus_service.py
   creating: dbus-opendtu-2.0.2/docs/
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.5.93_inverter-id-0.json
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.5.93_inverter-id-1.json
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.5.93_live.json
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.5.93_record-live.json
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.6.9_live.json
  inflating: dbus-opendtu-2.0.2/docs/ahoy_0.7.36_live_gap_in_inverter_sequence.json
  inflating: dbus-opendtu-2.0.2/docs/ahoy_pre_0.5.93.json
  inflating: dbus-opendtu-2.0.2/docs/opendtu_status.json
  inflating: dbus-opendtu-2.0.2/docs/opendtu_v24.2.12_inverter.json
  inflating: dbus-opendtu-2.0.2/docs/opendtu_v24.2.12_livedata_status.json
  inflating: dbus-opendtu-2.0.2/docs/tasmota_shelly_2pm.json
  inflating: dbus-opendtu-2.0.2/helpers.py
  inflating: dbus-opendtu-2.0.2/imports.py
  inflating: dbus-opendtu-2.0.2/install.sh
  inflating: dbus-opendtu-2.0.2/restart.sh
  inflating: dbus-opendtu-2.0.2/run_coverage.sh
   creating: dbus-opendtu-2.0.2/service/
   creating: dbus-opendtu-2.0.2/service/log/
  inflating: dbus-opendtu-2.0.2/service/log/run
  inflating: dbus-opendtu-2.0.2/service/run
  inflating: dbus-opendtu-2.0.2/tests.py
   creating: dbus-opendtu-2.0.2/tests/
  inflating: dbus-opendtu-2.0.2/tests/__init__.py
  inflating: dbus-opendtu-2.0.2/tests/test_dbus_opendtu.py
  inflating: dbus-opendtu-2.0.2/tests/test_dbus_service.py
  inflating: dbus-opendtu-2.0.2/tests/test_helpers.py
  inflating: dbus-opendtu-2.0.2/uninstall.sh
  inflating: dbus-opendtu-2.0.2/update.sh
  inflating: dbus-opendtu-2.0.2/version.txt
root@einstein:~# mv temp_dir/* /data/dbus-opendtu-main/
root@einstein:~# cp -R /data/dbus-opendtu-main/* /data/dbus-opendtu
root@einstein:~# chmod a+x /data/dbus-opendtu/*.sh
root@einstein:~# /data/dbus-opendtu/install.sh
grep: version.txt: No such file or directory
grep: version.txt: No such file or directory
root@einstein:~# /data/dbus-opendtu/restart.sh
root@einstein:~# rm main.zip
root@einstein:~# rm -r /data/dbus-opendtu-main/
```

`root@einstein:~# /data/dbus-opendtu/restart.sh`

mfreye, 2025-04-25:
```
#Possible Options for Log Level: CRITICAL, ERROR, WARNING, INFO, DEBUG, NOTSET
#To keep current.log small use ERROR
Logging=DEBUG
```

```
root@einstein:~# /data/dbus-opendtu/restart.sh
root@einstein:~# cat /var/log/dbus-opendtu/current | tai64nlocal > current.log
```

```

root@einstein:~# nano  current.log
  GNU nano 7.2                                                                  current.log
2025-04-25 13:04:44.198573500     config = getConfig()
2025-04-25 13:04:44.198581500              ^^^^^^^^^^^
2025-04-25 13:04:44.198583500   File "/data/dbus-opendtu/dbus_opendtu.py", line 19, in getConfig
2025-04-25 13:04:44.199716500     config.read(f"{(os.path.dirname(os.path.realpath(__file__)))}/config.ini")
2025-04-25 13:04:44.200659500   File "/usr/lib/python3.12/configparser.py", line 684, in read
2025-04-25 13:04:44.204998500     self._read(fp, filename)
2025-04-25 13:04:44.205824500   File "/usr/lib/python3.12/configparser.py", line 1049, in _read
2025-04-25 13:04:44.213190500     raise DuplicateSectionError(sectname, fpname,
2025-04-25 13:04:44.213676500 configparser.DuplicateSectionError: While reading from '/data/dbus-opendtu/config.ini' [line 89]: section 'INVERTER5' already exists
2025-04-25 13:04:44.596638500 Version: 2.0.2
2025-04-25 13:04:47.728323500 Traceback (most recent call last):
2025-04-25 13:04:47.728469500   File "/data/dbus-opendtu/dbus_opendtu.py", line 193, in <module>
2025-04-25 13:04:47.729896500     main()
2025-04-25 13:04:47.730289500   File "/data/dbus-opendtu/dbus_opendtu.py", line 160, in main
2025-04-25 13:04:47.731525500     config = getConfig()
2025-04-25 13:04:47.731920500              ^^^^^^^^^^^
2025-04-25 13:04:47.731982500   File "/data/dbus-opendtu/dbus_opendtu.py", line 19, in getConfig
2025-04-25 13:04:47.732886500     config.read(f"{(os.path.dirname(os.path.realpath(__file__)))}/config.ini")
2025-04-25 13:04:47.733739500   File "/usr/lib/python3.12/configparser.py", line 684, in read
2025-04-25 13:04:47.737279500     self._read(fp, filename)
2025-04-25 13:04:47.737809500   File "/usr/lib/python3.12/configparser.py", line 1049, in _read
2025-04-25 13:04:47.744648500     raise DuplicateSectionError(sectname, fpname,
2025-04-25 13:04:47.744656500 configparser.DuplicateSectionError: While reading from '/data/dbus-opendtu/config.ini' [line 89]: section 'INVERTER5' already exists
2025-04-25 13:04:48.364099500 Version: 2.0.2
2025-04-25 13:04:51.733144500 Traceback (most recent call last):
2025-04-25 13:04:51.733152500   File "/data/dbus-opendtu/dbus_opendtu.py", line 193, in <module>
2025-04-25 13:04:51.733155500     main()
2025-04-25 13:04:51.733156500   File "/data/dbus-opendtu/dbus_opendtu.py", line 160, in main
2025-04-25 13:04:51.736439500     config = getConfig()
2025-04-25 13:04:51.736447500              ^^^^^^^^^^^
2025-04-25 13:04:51.736450500   File "/data/dbus-opendtu/dbus_opendtu.py", line 19, in getConfig
2025-04-25 13:04:51.736453500     config.read(f"{(os.path.dirname(os.path.realpath(__file__)))}/config.ini")
2025-04-25 13:04:51.737435500   File "/usr/lib/python3.12/configparser.py", line 684, in read
2025-04-25 13:04:51.740859500     self._read(fp, filename)
2025-04-25 13:04:51.741003500   File "/usr/lib/python3.12/configparser.py", line 1049, in _read
2025-04-25 13:04:51.746390500     raise DuplicateSectionError(sectname, fpname,
2025-04-25 13:04:51.748521500 configparser.DuplicateSectionError: While reading from '/data/dbus-opendtu/config.ini' [line 89]: section 'INVERTER5' already exists
2025-04-25 13:04:52.251091500 Version: 2.0.2
2025-04-25 13:04:55.693441500 Traceback (most recent call last):
2025-04-25 13:04:55.693623500   File "/data/dbus-opendtu/dbus_opendtu.py", line 193, in <module>
2025-04-25 13:04:55.695267500     main()
2025-04-25 13:04:55.695668500   File "/data/dbus-opendtu/dbus_opendtu.py", line 160, in main
2025-04-25 13:04:55.697085500     config = getConfig()
2025-04-25 13:04:55.697501500              ^^^^^^^^^^^
2025-04-25 13:04:55.697569500   File "/data/dbus-opendtu/dbus_opendtu.py", line 19, in getConfig
2025-04-25 13:04:55.698596500     config.read(f"{(os.path.dirname(os.path.realpath(__file__)))}/config.ini")
2025-04-25 13:04:55.699225500   File "/usr/lib/python3.12/configparser.py", line 684, in read
2025-04-25 13:04:55.702769500     self._read(fp, filename)
2025-04-25 13:04:55.703353500   File "/usr/lib/python3.12/configparser.py", line 1049, in _read
2025-04-25 13:04:55.708062500     raise DuplicateSectionError(sectname, fpname,
2025-04-25 13:04:55.708243500 configparser.DuplicateSectionError: While reading from '/data/dbus-opendtu/config.ini' [line 89]: section 'INVERTER5' already exists
2025-04-25 13:04:56.172017500 Version: 2.0.2
2025-04-25 13:04:59.588423500 Traceback (most recent call last):
2025-04-25 13:04:59.588430500   File "/data/dbus-opendtu/dbus_opendtu.py", line 193, in <module>
2025-04-25 13:04:59.589704500     main()
[... 5 lines cut ...]
2025-04-25 13:04:59.594771500   File "/usr/lib/python3.12/configparser.py", line 684, in read
2025-04-25 13:04:59.598650500     self._read(fp, filename)
2025-04-25 13:04:59.604404500   File "/usr/lib/python3.12/configparser.py", line 1049, in _read
2025-04-25 13:04:59.609900500     raise DuplicateSectionError(sectname, fpname,
2025-04-25 13:04:59.610099500 configparser.DuplicateSectionError: While reading from '/data/dbus-opendtu/config.ini' [line 89]: section 'INVERTER5' already exists
2025-04-25 13:05:00.081713500 Version: 2.0.2
[... cut ...]
2025-04-25 13:05:03.136236500   File "/usr/lib/python3.12/configparser.py", line 1049, in _read
2025-04-25 13:05:03.140917500     raise DuplicateSectionError(sectname, fpname,
2025-04-25 13:05:03.141103500 configparser.DuplicateSectionError: While reading from '/data/dbus-opendtu/config.ini' [line 89]: section 'INVERTER5' already exists
2025-04-25 13:05:03.878700500 Version: 2.0.2
2025-04-25 13:05:06.947301500 Traceback (most recent call last):
2025-04-25 13:05:06.947470500   File "/data/dbus-opendtu/dbus_opendtu.py", line 193, in <module>
2025-04-25 13:05:06.949374500     main()

```

mfreye, 2025-04-25:
PS Danke henne49 für die Hilfe :)

0x7878, 2025-04-25:
Deine config ist nicht okay.

 section 'INVERTER5' already exists

Der Eintrag ist doppelt drin .

mfreye, 2025-04-25:
Vielen Dank.
Jetzt geht es wieder :)