henne49/dbus-opendtu #249: Script sends last value continusly when communication is aborted

SlapJackNpNp, 2025-05-12:
**Please Upgrade the Script to the latest version first**
Script is Up to date

**Describe the Problem**
Hello, i have the problem that ocassionaly the open dtu seem to get hung up or loose the wifi connection. Either way, if this happens the last value keeps sending constantly, leading to 800W PV at night for example. 

**Device you are using**
1. Which Venus OS are you using 3.50? -> Venus 3.60~72
2. Which Victron Device (Cerbo, Raspi, Multiplus II GX...)? -> Raspi and Cerbo
3. Which DTU (ahoy/opendtu/opendtuonbattery) -> openDTU
4. Which Template Device (Shelly, tasmota,...)? -> No Template only inverter

**Expected behavior**
If the DTU doesn't send data for a certain ammount of time, set the shown value to 0. Energy Values will get added as soon as the communication works again from what i saw. 

**Screenshots**

[image]

[image]

**Additional context**
I have this problem with AllianceApps openDTU and with Diamex DX-DTU and with a Raspi VenusOS and a Cerbo GX.

Saarblick, 2025-05-12:
Surprising, I realise the same issue over the last week.

SlapJackNpNp, 2025-05-12:
I can confirm that i can't acces the openDTU anymore in these situation. Usually i have to restart it to work again. 

I can provide a log the next time it happens. 

The energy is correct as soonnas it starts sending again, but if you look at the current data it looks wring and the vonsumption data which venusos calculates gets shown also wrong, leading to some confusion.

0x7878, 2025-05-12:
I think the problem is that our script isn't receiving any data from OpenDTU, so it doesn't update the values—there's simply nothing to update.

Henne is right: it doesn't affect the energy statistics or anything like that. This happens to me sometimes too. When AhoyDTU doesn't receive any new values from my HM-600, it just shows 3W constantly throughout the night. But the energy produced during the day is still reported correctly.

RiRomain, 2025-05-12:
I have the same issue, which cause Victron to export / store in the battery energy pulled from the grid, as it thinks the solar is still producing.
Sending 0 after no update from the inverter for X minutes could help with the issue.

Saarblick, 2025-05-12:
I don't has seen the issue in OpenDTU.
But I will check any restart in this timeframe..
I checked the MQTt value in IOBroker there I don't see the issue.. in the morning it was zero.
But I applied the history for this value in IOBroker.
I will look next morning.

SlapJackNpNp, 2025-05-12:
A Zimer chrcking for recived data would be great. If no data are recived for 2 or 3 minutes, showing the inverter as not connected or 0W would be great to see that something is not working correctly.

0x7878, 2025-05-12:
Logs would be helpful in any way. I could then look into the codebase.

SlapJackNpNp, 2025-05-12:
Sure, attached is a new log. I deleted the old log and rebooted the cerbo. There is not much power comming atm but it slowly went down from 36 to 34W and then i pulled the plug. In the log you can see that the script tries to connect to the dtu but gets no response. Yet the data in the cerbo keep showing 34W and status running

[image]

[current.txt](https://github.com/user-attachments/files/20166373/current.txt)

I hope this helps you?