henne49/dbus-opendtu #256: Problems by Installation the script by CerboGX

hubertusmantel, 2025-08-10:
**Please Upgrade the Script to the latest version first**
[Upgrade the Code](https://github.com/henne49/dbus-opendtu?tab=readme-ov-file#update-the-code)

**Last Upgrade was taken**

**Follow the Troubleshooting Section**
[Troubleshooting Section](https://github.com/henne49/dbus-opendtu#troubleshooting)
Please revert back to ERROR or CRITICAL, once debugging and troubleshooting is complete.
Please remove all confidential information first, before attaching the files. 

**Describe the Problem**
A clear and concise description of what the issue is.
**--> see screenshot below in screenshots**

**Device you are using** 
Steps to reproduce the behavior:
1. Which Venus OS are you using 3.50? **--> Venus Os 3.64** 
2. Which Victron Device (Cerbo, Raspi, Multiplus II GX...)? **--> CerboGX**
3. Which DTU (ahoy/opendtu/opendtuonbattery) --> 
4. Which Template Device (Shelly, tasmota,...)?

**Expected behavior**
A clear and concise description of what you expected to happen.
The script cannot be fully installed.

**Screenshot systeminformation openDTU**
[image]

**Screenshots Windows Power shell CerboGX**
Information about Installation skript in Windows Power shell

[image]
[image]
[image]

Completely remove the script with command rm -rf /data/dbus-opendtu is not possible.

schmidt3152, 2025-08-12:
Hello Henne, after entering the command this message appears.

root@192.168.188.20's password: 
Last login: Tue Aug 12 05:45:05 2025 from 192.168.188.38
root@einstein:~# mv temp_dir/* /data/dbus-opendtu
mv: can't rename 'temp_dir/Do you really wish to delete the log folder ': No such file or directory
mv: can't rename 'temp_dir/dbus-opendtu-2.0.2': No such file or directory
mv: can't rename 'temp_dir/kup': No such file or directory
mv: can't rename 'temp_dir/ll': No such file or directory
mv: can't rename 'temp_dir/mkdir: can't create directory 'temp_dir': File exists': No such file or directory
mv: can't rename 'temp_dir/r': No such file or directory
mv: can't rename 'temp_dir/rm -r ': No such file or directory
mv: can't rename 'temp_dir/root@einstein:~# unzip main.zip -d temp_dir': No such file or directory
mv: can't rename 'temp_dir/ve': No such file or directory
root@einstein:~# chmod a+x /data/dbus-opendtu/*.sh
chmod: /data/dbus-opendtu/*.sh: No such file or directory
root@einstein:~#

hubertusmantel, 2025-08-12:
Hello henne49,
yes it is a new installation.

After entering the commands i get this message back

[image]

Pressing the Enter key again results is ...

[image]

hubertusmantel, 2025-08-13:
Hello henne49,
I found the temp_dir using the command ls temp_dir and was then able to delete the temp_dir directory in CerboGX via Putty using the command rm -r temp_dir.
I then reinstalled the script and it has now been running for around 4 hours without any problems.

Thanks for your support

[image]

It is not possible to display the micro inverters separately in Victron's VRM portal?